beagleboard/bonescript #13: Add C function support

jadonk, 2018-04-25:
_From @jadonk on July 26, 2012 18:8_

It shouldn't be required to create node modules to add C functions that are compiled on the target. Need to merge in support for C routines that leverage Arduino-for-Userspace-Linux.

_Copied from original issue: jadonk/bonescript#27_

jadonk, 2018-04-26:
The Arduino-for-Userspace-Linux library seems to be largely superseded by MRAA and the now "native" support by create.arduino.cc.

anujdeshpande, 2018-04-30:
So does first class support with MRAA help us work with create.arduino.cc? 

I see that we are already [supported](https://github.com/intel-iot-devkit/mraa/blob/master/docs/beaglebone.md) - makes sense to add first class support? (looks like there are still [things](https://github.com/intel-iot-devkit/mraa/issues/749)  to do to get there)

jadonk, 2018-05-01:
It is still riddled with bugs to the best of my knowledge,  but this could increase the user base significantly.

vaishnav98, 2018-05-02:
@jadonk @anujdeshpande I have tried to implement external C function support through this example here : [https://github.com/vaishnav98/bonescript/tree/master/test/test-ffi_spi].can you please review it , i had the code tested on BeagleBone Black with
Linux beaglebone 4.9.88-ti-r111  and  BeagleBoard.org Debian Image 2018-03-05

I tried to implement the exported SPI functions to bonescript so that it can be called like this :
var b=require('bonescript')
b.openSPI(); 
...

but in this case the example works only when the example.js is in the same directory as the spi.js, and throws an error "File not Found spidev.so" , it seems like a path issue how can i solve it?can it be solved by adding the path to the .so file in LD_LIBRARY_PATH environment variable, if yes , how do i modify the makefile to accomodate this?