beagleboard/bonescript #15: Can't use 2 PWMs

jadonk, 2018-04-25:
_From @daguej on July 27, 2013 7:23_

If I try to use EHRPWM1A and EHRPWM1B:

```
b.pinMode('P9_14', b.OUTPUT);
b.pinMode('P9_16', b.OUTPUT);

b.analogWrite('P9_14', 0.5, 100);
b.analogWrite('P9_16', 0.5, 100);
```

...whichever is initialized second doesn't work.  If I look in /sys/devices/ocp.2,

```
# ls pwm*
pwm_test_P9_14.14:
driver  duty  modalias  period  polarity  power  run  subsystem  uevent

pwm_test_P9_16.15:
modalias  power  subsystem  uevent
```

_Copied from original issue: jadonk/bonescript#49_

jadonk, 2018-04-25:
I've confirmed that use of the PWM requires some specific period settings that the BoneScript library is violating. Here's the kernel log:
[10262.983541] ehrpwm 48302200.ehrpwm: Period value conflicts with channel 0
[10262.990960] pwm_test pwm_test_P9_16.16: pwm_config() failed
[10263.010070] pwm_test: probe of pwm_test_P9_16.16 failed with error -22
[10263.010171] bone-capemgr bone_capemgr.8: slot #10: Applied #2 overlays.

I believe that I'll need to autogenerate the .dts file to work around this limitation in the driver.

jadonk, 2018-04-25:
_From @daguej on August 20, 2013 3:16_

Thanks for looking in to this

jadonk, 2018-04-25:
Issue still exists in 0.2.4, but I know how to fix it.

jadonk, 2018-04-25:
_From @tovwiv on June 30, 2015 7:6_

any update on this? ran into this exact issue today

jadonk, 2018-04-25:
_From @adityapatadia on June 30, 2015 7:7_

May be you can try https://www.npmjs.com/package/octalbonescript

jadonk, 2018-04-25:
_From @NikolaMorena on September 29, 2015 11:41_

Workaround: use pins that are not connected with same EHRPWM module, e.g. P9_14 and P8_13

jrh21, 2020-06-02:
@jadonk was there any fix to this?