beagleboard/bonescript #17: Interchanging digital and analog functions

jadonk, 2018-04-25:
_From @jadonk on January 17, 2014 20:24_

It should be possible to interchange digital and analog functions for them to do something rational.

digitalRead on AIN: if > THRESHOLD then return HIGH, otherwise return LOW
analogRead on DIN: return 0 or 1 depending on digital state
digitalWrite on PWM: set duty cycle to 0 or 1
analogWrite on DOUT: if > THRESHOLD then set to HIGH, otherwise set to LOW

THRESHOLD = 0.5

_Copied from original issue: jadonk/bonescript#66_

jadonk, 2018-04-25:
_From @colinbes on February 6, 2014 13:30_

For digitalRead on AIN to reduce output chatter I'd advise adding a 'window' function to take care of noise on pin, for example.

```
if (value >= (threshold + window)) {
  return HIGH 
} else {
  if (value  < (threshold - window)) {
return LOW
}
```

This does leave a band gap of non determinable value which is actually normal in digital circuit design.

jadonk, 2018-04-25:
I can only return HIGH or LOW, so adding a window doesn't enable me to return any more interesting value. If there was some mechanism of sampling more often and retaining the previous value if the value was in the window, that makes some sense to act more like a digital input with a Schmitt trigger. In this case, I think a simple threshold is sufficient.

Also, be aware that the BeagleBone analog inputs are 1.8V, so driving them to 3.3V will damage them.

jadonk, 2018-04-25:
_From @colinbes on February 6, 2014 14:8_

My suggestion is to still only return HIGH or LOW but when checking the threshold to add a window function to stop chatter but if you can’t retain previous value (as default return) then this is not going to work.

If this was the case, then to be honest, if I was wanting to have such a function I’d roll my own and not use the library supplied function as it would result in too much chatter,  maybe it’s better left to the user to implement function where prior value can be retained?

~C

Internet Disclaimer

---

This message (including any attachments) contains confidential information intended for a specific individual and purpose, and may be protected by law. 
If you are not the intended recipient, you should delete this message and are hereby notified that any disclosure, copying, or distribution of this message, or the taking of any action based on it, is strictly prohibited.

---

vaishnav98, 2018-05-02:
@jadonk doesn't digitalWrite() on pwm work by default?