beagleboard/bonescript #29: Problem with GPIO / pinMode

jadonk, 2018-04-25:
_From @jukkaaakula on August 22, 2014 12:12_

Hi

I use 
                b.pinMode( outpin[j], b.OUTPUT , 7, 'pullup', 'fast', function (x) {} );

where outpin[j] is e.g. P8_31 or almost whatever. 

I get this:

/usr/local/lib/node_modules/bonescript/index.js:161
    if(typeof resp.err != 'undefined') {
                  ^
TypeError: Cannot read property 'err' of undefined
    at Object.f.pinMode (/usr/local/lib/node_modules/bonescript/index.js:161:19)
    at Object.event [as /stream](/home/debian/nodeSofta/requestHandlers.js:287:                                                                                                                     19)
    at route (/home/debian/nodeSofta/router.js:4:25)
    at IncomingMessage.<anonymous> (/home/debian/nodeSofta/server.js:18:13)
    at IncomingMessage.EventEmitter.emit (events.js:92:17)
    at _stream_readable.js:920:16
    at process._tickCallback (node.js:415:13)

Actually even for simpler versions I get the same message. E.g. for

```
            b.pinMode( outpin[j], b.OUTPUT );
```

Beaglebone version is 2.4. OS was Debian with node.js 10.*

_Copied from original issue: jadonk/bonescript#93_

jadonk, 2018-04-25:
_From @adityapatadia on September 15, 2014 7:39_

Try https://github.com/theoctal/octalbonescript It is fork of bonescript with a lot of improvements.

vaishnav98, 2018-05-06:
This issue does not occur with the latest version , tested on BeagleBone Black with
Linux beaglebone 4.9.88-ti-r111 and BeagleBoard.org Debian Image 2018-03-05