beagleboard/bonescript #47: Unable to unload conflicting slot

jadonk, 2018-04-25:
_From @psiphi75 on December 11, 2015 4:37_

I am running bonescript 0.2.5 (the latest) on a Beaglebone Black RevC.  Running an up-to-date Debian 7 (7.9??) and kernel as follows:

```
Linux beaglebone 3.8.13-bone79 #1 SMP Tue Oct 13 20:44:55 UTC 2015 armv7l GNU/Linux
```

I get the following error when running the code listed here (http://beagleboard.org/support/BoneScript/ServoMotor/):

```
Unable to unload conflicting slot: Write to CapeMgr slots failed: Error: EEXIST, file already exists
```

I have determined that the file that "exists" in the error above is:

```
/sys/devices/bone_capemgr.9/slots
```

The code that I have provided a link to is:

``` JavaScript
var SERVO = 'P9_14';
var duty_min = 0.03;
var position = 0;
var increment = 0.1;

b.pinMode(SERVO, b.OUTPUT);
updateDuty();

function updateDuty() {
    // compute and adjust duty_cycle based on
    // desired position in range 0..1
    var duty_cycle = (position*0.115) + duty_min;
    console.log(duty_cycle);
    b.analogWrite(SERVO, duty_cycle, 60, scheduleNextUpdate);
    console.log("Duty Cycle: " +
        parseFloat(duty_cycle*100).toFixed(1) + " %");
}

function scheduleNextUpdate() {
    // adjust position by increment and
    // reverse if it exceeds range of 0..1
    position = position + increment;
    if(position < 0) {
        position = 0;
        increment = -increment;
    } else if(position > 1) {
        position = 1;
        increment = -increment;
    }

    // call updateDuty after 200ms
    setTimeout(updateDuty, 200);
}
```

Note, it took a bit of debugging to get the messages above.  The actual error I get is:

```
/usr/local/lib/node_modules/bonescript/src/my.js:230
            callback(resp);
            ^
TypeError: undefined is not a function
    at onUnloadSlot (/usr/local/lib/node_modules/bonescript/src/my.js:230:13)
    at unloadSlot (/usr/local/lib/node_modules/bonescript/src/my.js:219:13)
    at onWriteSlots (/usr/local/lib/node_modules/bonescript/src/my.js:193:43)
    at onReadSlots (/usr/local/lib/node_modules/bonescript/src/my.js:182:13)
    at onFindCapeMgr (/usr/local/lib/node_modules/bonescript/src/my.js:157:9)
    at Object.exports.load_dt (/usr/local/lib/node_modules/bonescript/src/my.js:140:5)
    at onDTBOExists (/usr/local/lib/node_modules/bonescript/src/my.js:317:26)
    at onDTBOExistsTest (/usr/local/lib/node_modules/bonescript/src/my.js:264:13)
    at Object.exports.create_dt (/usr/local/lib/node_modules/bonescript/src/my.js:259:9)
    at Object.exports.setPinMode (/usr/local/lib/node_modules/bonescript/src/hw_capemgr.js:102:12)
```

_Copied from original issue: jadonk/bonescript#111_

vaishnav98, 2018-05-06:
This issue does not occur with the latest version , tested on BeagleBone Black with
Linux beaglebone 4.9.88-ti-r111 and BeagleBoard.org Debian Image 2018-03-05

jadonk, 2018-05-21:
Latest probably uses hw_mainline.js. Keeping these hw_oldkernel.js, hw_capemgr.js and hw_universal.js was meant to try to support old kernels. I'm not sure if it is viable. You'd need to get old kernels to test these functions. Considering will-not-fix, but would need to explicitly state a series of supported kernels that were reasonable.