beagleboard/bonescript #49: How to remotely deploy code to the BeagleBone?

jadonk, 2018-04-25:
_From @bennyn on March 5, 2017 18:45_

Hello, I am new to the BeagleBone community and I would like to know if it is possible to deploy JavaScript code remotely to the BeagleBone?

Until now I always coded on the BeagleBone (using the Cloud9 environment) itself. But I would like to code on my personal computer and then deploy it from there to the BeagleBone. Is this possible?

I know that the [Tessel](https://tessel.io/) microcontroller supports this way of coding by running `tessel run my-app.js`.

_Copied from original issue: jadonk/bonescript#148_

jadonk, 2018-04-25:
_From @jgtaylor on March 6, 2017 11:19_

i do my dev on my laptop and push to github, then pull from the BBblack.

You could also do a rsync or an scp from your dev host to your BBblack. If you install something like pm2 and use it to watch wherever you're putting your project, it will restart the app if it detects changes, so that after each update iteration, it automatically runs the app.

another possibility (but is kind of overkill) is to run a git server on the BBblack and just push your code there.

anujdeshpande, 2018-04-30:
Makes sense to add a `over the internet` option? Using something like gists from GitHub? Not sure how we'd get the output back? Maybe using another gist?  (not sure if this makes sense for us, but would love to have Particle.io like functionality)

jadonk, 2018-05-21:
I did something like this at one point. Since BoneScript creates an RPC server.

https://github.com/beagleboard/bonescript/blob/7e1e6ce67a9d48a5f454b6d11e34e2dfbef069d2/test/TODO/remote_bonetest.js

BTW, this really needs some security. That should trump some of this other development. Just some kind of secure session creation when using the web server.

vaishnav98, 2018-05-21:
^Sure , I will go through it soon