beagleboard/bonescript #52: digitalWrite issue (conflict with analogWrite?)

udoom, 2018-08-30:
hello,

this code throws an error on all my BBblack (since bonescript 0.7.0 update)  : 
```
b.detachInterrupt('P9_29');`
b.pinMode('P9_29', b.OUTPUT);
b.digitalWrite('P9_29', b.LOW);

b.detachInterrupt('P9_21');
b.pinMode('P9_21', b.OUTPUT);
b.digitalWrite('P9_21', b.LOW);  // <-------- here
```

the second b.digitalWrite throws this error : 

`analogWrite: P9_21 requires pwm EHRPWM0B but it is already in use by P9_29`

it seems to be caused by this portion of code in "/usr/local/lib/node_modules/bonescript/src/index.js" in digitalWrite function : 
```
        var gpioEnabled = (7 == f.getPinMode(pin).mux); //check whether pin set as gpio
        if (!gpioEnabled) {
            winston.debug([pin.key, value] + ' set as ANALOG_OUTPUT modifying duty cycle according to value');
            f.analogWrite(pin, value, 2000, myCallback); //write duty cycle as per value
            return (true);
        }
```

because if we inspect "f.getPinMode(pin)" it shows no mux property : 

for P9_29 : 
```
`{ pin: 'P9_29',
  name: 'SPI1_D0',
  options:
   [ 'mcasp0_fsx',
     'ehrpwm0B',
     'NA',
     'spi1_d0',
     'mmc1_sdcd',
     'pr1_pru0_pru_r30_1',
     'pr1_pru0_pru_r31_1',
     'gpio3_15' ],
  gpio: { active: true, direction: 'out', allocated: true } }`
```

for P9_21 :
```
`{ pin: 'P9_21',
  name: 'UART2_TXD',
  options:
   [ 'spi0_d0',
     'uart2_txd',
     'i2c2_scl',
     'ehrpwm0B',
     'pr1_uart_rts_n',
     'pr1_edio_latch_in',
     'EMU3',
     'gpio0_3' ],
  gpio: { active: true, direction: 'out', allocated: true } }`
```

if I comment this code it works : 
```
        if (!gpioEnabled) {
            winston.debug([pin.key, value] + ' set as ANALOG_OUTPUT modifying duty cycle according to value');
            f.analogWrite(pin, value, 2000, myCallback); //write duty cycle as per value
            return (true);
        }
```

if you need more informations, tell me

vaishnav98, 2018-09-12:
Hi @wam31  I have tried to fix this issue , could you also please test and confirm whether it is fixed?

wam31, 2018-09-12:
hello @vaishnav98 , your patch fix the issue
thank's !

do you plan to release bonescript 0.7.1 ?

vaishnav98, 2018-09-12:
Glad to hear that :) , I believe @jadonk  will be including these patches in the next release soon after review.

lukdoo, 2019-02-21:
hello @jadonk @vaishnav98 ,
why the fixeing PR [#180](https://github.com/jadonk/bonescript/pull/180) is still opened ?
you don't want to integrate these fixes in a release ?

thx

closed 2019-08-15 by jadonk