beagleboard/bonescript #53: analogWrite slow when writing the value 0

Aidan275, 2018-09-09:
Hi, 

Since upgrading from bonescript 0.6.3 to 0.7.0 I have found that the analogWrite method takes 2-5 seconds to write the value 0 after writing any other value greater than 0. 
There is also a large delay when writing to any other value greater than 0 from 0.

Please see the example below which illustrates the issue.

The following code was run after boot on a BeagleBone Black in node v8.11.4 and with kernel 4.14.49-ti-r54.
```
const b = require("bonescript");

b.pinMode('P8_19', b.ANALOG_OUTPUT);

function write(val) {
    const start = Date.now();
    b.analogWrite('P8_19', val, 2000, x => {
        if (x.err) {
            console.log(x.err);
        } else {
            console.log((Date.now() - start) + "ms");
        }
    });
}

write(0);
write(0);
write(0);
write(0);
write(1);
write(1);
write(1);
write(1);
write(1);
write(0);
write(0);
write(0);
write(0);
write(1);
write(1);
write(1);
write(1);
write(1);
```
The output from this code is:
```
> const b = require("bonescript");
>
> b.pinMode('P8_19', b.ANALOG_OUTPUT);
> 
> function write(val) {
...     const start = Date.now();
...     b.analogWrite('P8_19', val, 2000, x => {
.....         if (x.err) {
.......             console.log(x.err);
.......         } else {
.......             console.log((Date.now() - start) + "ms");
.......         }
.....     });
... }
> 
> write(0);
39ms
> write(0);
11ms
> write(0);
7ms
> write(0);
11ms
> write(1);
13ms
> write(1);
111ms
> write(1);
6ms
> write(1);
6ms
> write(1);
2ms
> write(0);
4481ms
> write(0);
4437ms
> write(0);
3602ms
> write(0);
3172ms
> write(1);
3681ms
> write(1);
2ms
> write(1);
2ms
> write(1);
2ms
> write(1);
3ms
```
If you exit nodejs after this and then run it again, the first write(0) calls also have a large delay (sometimes 15+ seconds for the first one).

Please let me know if you need anymore infromation from me. 

Thanks

riptho, 2018-09-12:
Also having this issue.

> Hi,
>
> Since upgrading from bonescript 0.6.3 to 0.7.0 I have found that the analogWrite method takes 2-5 seconds to write the value 0 after writing any other value greater than 0.
> There is also a [...]

vaishnav98, 2018-09-12:
Hi @Aidan275  @riptho  I have tried to fix this issue , could you also please test and confirm whether it is fixed?

closed 2018-09-14 by Aidan275

vaishnav98, 2018-09-14:
Thank you @Aidan275 :)