beagleboard/bonescript #54: Dual outputs and locked PWM on the same EHRPWM module

Aidan275, 2018-10-04:
Hi,

When writing 1 using analogWrite to a PWM pin, the associated pin that is connected on the same EHRPWM module also outputs 1.

Trying to write 0 to this associated pin doesn't change the output and it can only be set to 0 if we use analogWrite to set it to 1 first. 

The example below will result in P9_14 and P9_16 both outputting 1.

```
const b = require('bonescript');
b.pinMode('P9_14', b.ANALOG_OUTPUT);
b.analogWrite('P9_14', 1, 2000);
```
I've only tested the following pins so far and they all appear to have the same issue.
- P8_13 and P8_19
- P9_14 and P9_16
- P9_21 and P9_22

I found a similar issue already posted here #15 but we are unable to use the suggested workaround of using pins that aren't connected on same EHRPWM module.

Thanks, 
Aidan

Aidan275, 2018-10-09:
Hi @jadonk, 

You mentioned in #15 that you know how to fix this issue, I know you must be busy but if you could please give me any info on the cause of the issue and point me in the right direction so that I can attempt to fix it I would really appreciate it.

Thanks,
Aidan

riptho, 2018-11-18:
I am also having this issue. Please advise.